adafruit/Adafruit_CircuitPython_HTTPServer #6: Documentation: `status` in response cannot be a tuple

nicolasff, 2022-06-29:
Hello,

I started using this library today and followed the documentation describing how to set the status code when responding to a request. The triple-quoted docstring for `HTTPResponse.__init__` describes `status` as:
>         :param tuple status: The HTTP status code to return, as a tuple of (int, "message").
>          Common statuses are available in `HTTPStatus`.

while the type hint for the constructor argument defines it as `status: tuple = HTTPStatus.OK`. The name `tuple` is recognized by the documentation generator, which creates a link to the Python.org documentation for the [tuple type](https://docs.python.org/3/library/stdtypes.html#tuple).

Passing a tuple does not actually work here, and causes the response to be badly formatted. The value given to `status` is saved in `self.status`, then passed to `self._send_response`, which in turn calls `self._send_bytes` with the value produced by `self._HEADERS_FORMAT.format`: https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/67c3ac3b76dd419a90558460b727a07b2169351f/adafruit_httpserver.py#L232-L236

This `format` method is `str.format`, since `self._HEADERS_FORMAT` is just a string: https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/67c3ac3b76dd419a90558460b727a07b2169351f/adafruit_httpserver.py#L179-L185

Passing a tuple as documented – e.g. `(200, 'OK')` causes it to be rendered at the position of the first `{}` in the format string, making the response look like this:
```
HTTP/1.1 (200, 'OK')
Content-Type: text/plain
Content-Length: 5
Connection: close

hello
```

This is not a valid HTTP response, so when `curl` receives it it fails with
```
* Unsupported HTTP version in response
* Closing connection 0
```

The docs do suggest to use the [common values](https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/67c3ac3b76dd419a90558460b727a07b2169351f/adafruit_httpserver.py#L53-L58) pre-defined as static fields in `HTTPStatus`, which aren't tuples but `HTTPStatus` instances. They are only provided for status code `200`, `404`, and `500`, so it's likely that users would want to provide other values and try to use tuples like `(403, 'Forbidden')` for example.
`HTTPStatus` instances are rendered correctly in the format string because the class specifically defines its string representation with a `__str__` method: https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/67c3ac3b76dd419a90558460b727a07b2169351f/adafruit_httpserver.py#L49-L50

I'm not sure whether this should be best handled as purely a documentation change – no longer suggesting `tuple` – or as a code change, actually supporting `tuple` and rendering it correctly, maybe even raising an exception if a tuple is provided that is not made of an int and a string. I'm far from an authority on the matter, but it seems more _Pythonic_ to me to actually support tuples rather than require an `HTTPStatus` object or a protocol-formatted string to directly inject into the response.

Thanks for this library!

michalpokusa, 2022-11-19:
Hi, I seems like my PR #25 fixes that problem, as it changes `tuple` to `HTTPStatus` instance everywhere except `HTTPResponse.__init__`. I also agree, that passing `tuple` is more _Pythonic_, so I made it possible to pass is to the constructor as it will be converted to `HTTPStatus` instance.

Please try the latest version of [refactor-and-additional-features branch](https://github.com/michalpokusa/Adafruit_CircuitPython_HTTPServer/tree/refactor-and-additional-features) and let me know whether it fulfills this Issue.

nicolasff, 2022-11-19:
Hi @michalpokusa, thanks for the update!

Your change with the type declaration and conversion looks promising.
```python
self.status = status if isinstance(status, HTTPStatus) else HTTPStatus(*status)
```

This is a pretty large PR, and although I see that someone from Adafruit has taken notice I just hope that they'll be ok with a change of this size and will have someone review it. Did you sync up with them on Discord, maybe? I've seen this work well in the past for non-trivial contributions. If you haven't I'd suggest discussing it there to get more direct feedback from Adafruit employees who work on the CircuitPython repos.

I'll follow your PR, in the meantime I think it still makes sense to keep this issue open for anyone who would run into this problem with the current release, and to close it once the fix is in.

michalpokusa, 2022-11-19:
Thanks for the advice regarding Discord, will definitely contact them.
I agree that keeping the Issue open for now is a good idea, if they decide to merge the PR then it will close automatically.

dhalbert, 2022-11-19:
I, as the original author of this, as I mentioned in #17, am quite interested in a revamp. I don't care about keeping it upward compatible. The current version number 0.x.0 reflects that. Feel free to rationalize and improve the API and make it better for the future.

closed 2022-11-22 by dhalbert