adafruit/Adafruit_CircuitPython_HTTPServer #18: no response to http requests on pico w /CP8 Beta

domi-a, 2022-10-23:
tried circuitpython 8 beta 1/2/3 but the HTTPServer is not usable on pico w.

minimal examples like all three from this very lib: https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/tree/main/examples
are not repsonding to simple browser or curl requests
> ERR_CONNECTION_REFUSED (like about every 50. request does in fact get a response)

then i read this, and he is absolutly right: https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/issues/14#issuecomment-1283293120
replacing the .mpy to source .py and changing L:334 to
`self._sock.setblocking(True)` 
will make the HTTPServer respond every time
https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/ba2da4352290b880592ae5ab5db83ee50c045244/adafruit_httpserver.py#L334

not sure if this is related to this issue, https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/issues/9 wich dies belong to this MR https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/pull/10/commits/bcabd1d924b9845ce4e9fc03788f1d7ebafd53a7 wich originally does change this very line to the opposite that makes the lib work for me and @justbuchanan.

sadly this is not a real solution, since it is now waiting/blocking to a incoming request and no other magic can happen while waiting :-(

anecdata, 2022-10-23:
When I originally (briefly) tested Pico W with http_server it worked, must have gotten lucky. But more recently, I'm seeing the same behavior with a simple asyncio http_server ...successful connections are rare. I wonder if there is less buffering in the internals on the Pico W. (On the bright side, the asyncio http_server occasionally hardfaults on esp32-s2, but not on Pico W.)

anecdata, 2022-10-25:
Should be fixed now with #19

closed 2022-10-25 by anecdata

domi-a, 2022-10-25:
> Should be fixed now with #19

wonderful. thank you.
i testet the change locally and it is now working as expected.