adafruit/Adafruit_CircuitPython_HTTPServer #28: Content-Length and multibyte characters in HTTPResponse

jun2sak, 2022-12-18:
I am building a tiny web server using adafruit_httpserver.server and circuitpython 8.0.0-beta.5 on ESP32-S2.
When multibyte characters (Japanese) are included in response body, the content-length header value received by a web client looks shorter, and some tailing characters in the body are missing.

```
page_html = """
<html>
  <head><title>test</title></head>
  <body>
   some text in multibyte language here..
  </body>
</html>
"""

@server.route("/test")
def base(request):
  return HTTPResponse(content_type='text/html;charset=UTF-8',body=page_html)
```

```
$ curl -v http://192.168.xx.xx/test
 :
< HTTP/1.1 200 OK
< Content-Length: 117
< Content-Type: text/html;charset=UTF-8
< Connection: close
<
* Excess found in a read: excess = 10, size = 117, maxdownload = 117, bytecount = 0
(html response are shown, but the last several characters in body are missing)
```

Looking into adafruit_httpserver/response.py, content-length is calculated as len(body), as in,
`response_headers.setdefault("Content-Length", content_length or len(body))`
and the whole response are sent after converted into bytes.
If I replace it with `len(body.encode("utf-8"))`, the above trouble disappears, but I'm not sure this modification is right.

[response.py#L78](https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/11ae0bc5f5a15dd26305006c95fbdcb46b17d61e/adafruit_httpserver/response.py#L78)

michalpokusa, 2022-12-19:
I can confirm that this is a solution to the problem, this is a leftover from the time when I was preparing code to return images as bytes, which I abandoned later. In the end the whole `response` is being `.encode("utf-8")` so this seems to be the correct fix for the problem. Currently I am working on overhaul of headers handling. I will include a fix for that in my next PR.

jun2sak, 2022-12-22:
> @jun2sak would you mind testing out PR 29? You can get the `.py` files from the PR "Files Changed" tab, or the `.mpy` files from the "Checks" tab.

In my environment, the problem I reported is resolved with the `.py` files from the PR 29.

jun2sak, 2022-12-22:
Using PR 29 files, I found another problem.

When I access a page that does not exist in the CircuitPython http server, as in `$ curl http://192.168.xx.xx/noexists`, I got an error in the server REPL console:
```
[srv.py]
@server.route("/test")
def base(request):
  return HTTPResponse(content_type='text/html;charset=UTF-8',body=page_html)

server.serve_forever(str(wifi.radio.ipv4_address))
```
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "srv.py", line 31, in <module>
  File "/lib/adafruit_httpserver/server.py", line 75, in serve_forever
  File "/lib/adafruit_httpserver/server.py", line 177, in poll
  File "/lib/adafruit_httpserver/response.py", line 110, in send
  File "/lib/adafruit_httpserver/response.py", line 131, in _send_response
  File "/lib/adafruit_httpserver/response.py", line 78, in _construct_response_bytes
AttributeError: 'NoneType' object has no attribute 'setdefault'
```
Using the original version, I got just 
```
$ curl http://192.168.xx.xx/noexists
404 Not Found /noexists
```
and no error in REPL console.

PR #29 (2023-01-02): Case insensitive HTTPHeaders, HTTPResponse context manager and some fixes
Added separate class for HTTP headers, that allows accessing headers using case-insensitive names.

Also, changed some docstrings the were unclear and one that was misleading as it was inconsistent with the code.

Changed the order of Docs References, so that the most "important" ones are on the top, with enums and dataclass-like classes below.

Updated library version to __1.1.0__ in sphinx configuration file.

Fixes #26
Fixes #28
closed 2023-01-02 by dhalbert