adafruit/Adafruit_CircuitPython_HTTPServer #31: HTTP Requests from Chrome block the server

yousafs, 2023-01-07:
There seems to be a problem handling requests from Chrome. Chrome locks the http server for a minute, blocking any other requests from being handled - other than requests from the same Chrome tab. A minute after the last request from Chrome, the server will start handling other requests again.

If you start a server, it will handle requests correctly from clients like node-red, firefox etc. (anything other than Chrome!). 
If you are looping with server.poll() the loop will run correctly, but will freeze for 60 seconds after the Chrome request.
An example from an ESP32-S3 I am using:

```
pool = socketpool.SocketPool(wifi.radio)
server = HTTPServer(pool)
server.request_buffer_size = 2048
server.socket_timeout = 1
server.start(str(wifi.radio.ipv4_address))

@server.route("/")
def base(request: HTTPRequest):
    with HTTPResponse(request, content_type=MIMEType.TYPE_TXT, chunked=False) as response:
        try:
            print("REQUEST: /")
            response.send("HELLO")
        except OSError:
            response.send("ERROR")

while True:
    try:
        led.value = True
        time.sleep(0.05)
        led.value = False
        time.sleep(0.1)
        server.poll()
        
    except OSError as error:
        print(error)
        continue
```

As soon as you hit the server with Chrome, the server responds but stops the loop and stops handling requests from other clients. If you make another request from chrome in the same tab, it will be handled, and seemingly at that time pending requests from other clients can sometimes be handled but the loop will freeze again right after.

Making a request to an undefined path from Chrome freezes the loop also - you get the 404 in chrome and then server loop freezes - whatever code you put in your handler does not fix this problem.

A minute after the last Chrome request, the loop restarts and the server functions normally.

If you quit Chrome while the loop is locked, the loop restarts immediately.

I've tried a bunch of things to fix this, including closing the connection in the handler, sending various keep-alive responses to drop the connection - none work.

I think this needs an SDK level fix!

yousafs, 2023-01-07:
Sorry, should have given my board/version:

`Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3`

anecdata, 2023-01-07:
Can you show the `request.headers` Chrome is sending?

yousafs, 2023-01-07:
Sure! Here is Chrome:

```
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Accept-Encoding: gzip, deflate
Accept-Language: en-GB,en-US;q=0.9,en;q=0.8
Cache-Control: max-age=0
Connection: keep-alive
Host: 192.168.99.6
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/108.0.0.0 Safari/537.36
```

and to make the point I think it is Chrome specific - here is the same request in Firefox:

```
Host: 192.168.99.6
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:106.0) Gecko/20100101 Firefox/106.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,*/*;q=0.8
Accept-Language: en-GB,en;q=0.5
Accept-Encoding: gzip, deflate
Connection: keep-alive
Upgrade-Insecure-Requests: 1
```

michalpokusa, 2023-01-07:
I tried the code you provided on ESP32-S2 TFT as it is the only WiFi board I have, works fine. Have you tried it on any other board?

yousafs, 2023-01-07:
Thanks so much for trying 👍
No I have not tried it on any other board (dont have another) happy to buy another one and try it ...

michalpokusa, 2023-01-07:
There are some things that come to my mind as the probable reason for that behavior.
The key is to find an exact line in code that blocks.

If it is ok for you we can try to find a cause of this on Discord call.

yousafs, 2023-01-07:
> There are some things that come to my mind as the probable reason for that behavior, mostly browser related. The key is to find an exact line in code that blocks.
> 
> If it is ok for you we can try to find a cause of this on Discord call.

Sure happy to do that - bear in mind that it blocks even if you hit a non-existent url (i.e. no obvious line of code executes?)
Just about to meet a pal (UK time - hence beer o'clock) but if you let me know how to do it I can do a discord call tomorrow?

michalpokusa, 2023-01-07:
By line of code I also meant the line in library code.

I'm located in Poland so there is only 1h diffrence between us. Tomorrow works for me, I'm flexible with time.
My Discord Michał Pokusa#2886, please add/message me so we can schedule a call for tomorrow.

yousafs, 2023-01-08:
> EDIT 1: First thing I would do is placing some print statements in adafruit_httpserver/server.py, you probably have a .mpy file so debugging it is not really possible, try downloading a .py version of library. Using these prints it should be possible to pinpoint where the block occurs.

So I did as you suggested and sprinkled print statements into server.py, the board pauses for 60 seconds in the _receive_header_bytes() function, specifically, during sock.recv_into() ...

```
    def _receive_header_bytes(
        self, sock: Union["SocketPool.Socket", "socket.socket"]
    ) -> bytes:
        print("Receive Header Bytes")
        """Receive bytes until a empty line is received."""
        received_bytes = bytes()
        while b"\r\n\r\n" not in received_bytes:
            try:
                print("Try Receive Header Bytes")
                length = sock.recv_into(self._buffer, len(self._buffer)) <------ pauses here
                print("Received Header Bytes")
                received_bytes += self._buffer[:length]
            except OSError as ex:
                print("Exception OS")
                if ex.errno == ETIMEDOUT:
                    break
            except Exception as ex:
                print("Exception Other")
                raise ex
        return received_bytes
```

I noticed that even with a request from Firefox, the code can linger there for a few seconds, but it does not pause the way that Chrome does.

michalpokusa, 2023-01-08:
Together with @yousafs we did some testing and came to the following conclusion:

- This problem does not appear on CircuitPython 7.3.3, only on 8.0.0
- Problem exists on both ESP32-S2 and ESP32-S3
- The core problem seems to be connected to probably incorrect behaviour of timeout on socket, despite setting it it is simply ignored, which scauses blocking
- On CP 7.3.3 after processing request program loops `.pool()`, on CP 8.0.0 most of the time it gets stuck on `sock.recv_into()`
- Serial connection seems to be blocked when stuck on `sock.recv_into()`

Separate issue will be opened on CP's repository.

yousafs, 2023-01-10:
Thanks Michal.
FWIW I think this will be a more robust fix across a range of feathers, rather than relying on the underlying socket library being fully implemented on every board.

michalpokusa, 2023-02-09:
Following the release of the stable CP 8.0.0 I tested whether this issue is still present.
Blocing is no longer the issue, the https://github.com/adafruit/circuitpython/pull/7455 fixed it a while ago, thanks @dhalbert!

I believe this issue can be closed as resolved.

anecdata, 2023-02-09:
@yousafs This should be resolved, you can re-open if you still have the issue.

closed 2023-02-09 by anecdata