adafruit/Adafruit_CircuitPython_HTTPServer #44: Pico W stuck in an infinite loop

foxy82, 2023-04-03:
If I do the GET request too quickly on a PICO W it gets stuck in an infinite loop.

I took inspiration from this issue on how to debug: https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/issues/41

When I add `print(bytes_sent, bytes_to_send)` to the response.py _send_bytes method I can see it gets stuck in an infinite loop.

Hard to capture in text as the board locks up and so does Mu serial connection I took a video and I've attached an image from the point where the issue happens.

code.py at the moment is:

```
import socketpool
import wifi
import os
from adafruit_httpserver.mime_type import MIMEType
from adafruit_httpserver.request import HTTPRequest
from adafruit_httpserver.response import HTTPResponse
from adafruit_httpserver.server import HTTPServer

if not wifi.radio.ipv4_address:
    ssid = os.getenv('WIFI_SSID')
    print(f"Connecting to Wifi: {ssid}")
    wifi.radio.connect(ssid, os.getenv('WIFI_PASSWORD'))
    print(f"Connected: IP address is {wifi.radio.ipv4_address}")

pool = socketpool.SocketPool(wifi.radio)
server = HTTPServer(pool)
server.socket_timeout = 0.25

count = 0
poll_count = 0

@server.route("/")
def base(request: HTTPRequest):
    print(request.path, request.connection)
    """
    Serve the default index.html file.
    """
    with HTTPResponse(request, content_type=MIMEType.TYPE_HTML) as response:
        response.send_file("index.html")

@server.route("/MUTE")
def mute(request: HTTPRequest):
    print(request.path, request.connection)
    """
    Serve the default index.html file.
    """
    try:
        global count
        count = count + 1
        print(f"{count} sending")
        # device.send(0xE2)
        with HTTPResponse(request, content_type=MIMEType.TYPE_HTML) as response:
            response.send_file("index.html")
        print(f"{count} sent")
    except Exception as e:
        print(e)

print(f"Listening on http://{wifi.radio.ipv4_address}:80")
# Start the server.
server.start(str(wifi.radio.ipv4_address))
while True:
    try:
        poll_count = poll_count + 1
        if(poll_count % 10_000 == 0): 
            print(f"Polling {poll_count}")
        server.poll()
        if(poll_count % 10_000 == 0): 
            print(f"Poll Done {poll_count}")
    except Exception as error:
        print(error)
        raise error

```

and index.html is:

```
<!DOCTYPE html>
<html lang="en">
<head>
  <title>REMOTE</title>
  <meta charset="utf-8">
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="stylesheet" href="https://cdn.jsdelivr.net/npm/bootstrap@4.0.0/dist/css/bootstrap.min.css" integrity="sha384-Gn5384xqQ1aoWXA+058RXPxPg6fy4IWvTNh0E263XmFcJlSAwiGgFAW/dAiS6JXm" crossorigin="anonymous">
</head>
<body class="bg-dark">
    <div class="container-sm bg-dark rounded text-center">
            <br>
            <div class="row">
                <div class="col-1">
                    <div class="text-light">
                      <a class="btn brn-dark btn-lg text-light btn-outline-primary" href="/MUTE" role="button">&nbsp;&#128264;&nbsp;</a>
                    </div>
                </div>
            </div>
    </div>
</body>
</html>
```

[image]

foxy82, 2023-04-03:
Ok so pico w gets stuck in an infinite loop in _send_bytes because we get an error code 32 EPIPE

I've got something that fixes it for me I'll raise a PR for review.

michalpokusa, 2023-04-03:
I do not own a Pico W, but from the code I see a couple of things.

You set `socket_timeout` to `0.25`, try commenting out this line or even setting it to e.g. `2`, maybe Pico W is not fast enought to process that many requests in that short time. By default it is set to `1`, in the past there were some problems with it, but after testing it seemed like a reasonable amount. 

I am unable to replicate the issue with ESP32S2 TFT Feather, maybe there is something wrong with the sockets on Pico's version of CP?

_PS:_
You shouldn't need to route `"/"` to `index.html`, server should do it automatically (#38). Also it is not necessary to provide `content_type` when using `send_file`, server determines it from the file's name, the value passed to constructor is not used.

foxy82, 2023-04-03:
The `0.25` was for testing to see if I could get it to recover - issue happens with it set to `1`. I think the pico w just throws a broken pipe if it gets overloaded. So not swallowing this (as per the PR) fixes the issue and it recovers on the next poll.

dhalbert, 2023-04-03:
Tagging @jepler for interest

foxy82, 2023-04-03:
> The `0.25` was for testing to see if I could get it to recover - issue happens with it set to `1`. I think the pico w just throws a broken pipe if it gets overloaded. So not swallowing this (as per the PR) fixes the issue and it recovers on the next poll.

As it is the send from Pico W -> Chrome I guess it could be somehow Chrome -> Pico W wifi breaks the connection and the Pico W is just genuinely reporting a Broken Pipe.... 

I get the same error using FireFox as well - testing in case it was a browser specific thing.

FoamyGuy, 2023-04-24:
Resolved by: #45

closed 2023-04-24 by FoamyGuy