adafruit/Adafruit_CircuitPython_HTTPServer #50: Create an httpserver_simpletest example

jrrickerson, 2023-04-25:
Create a bare minimum example that proves the HTTPServer is running on the device, akin to a "Hello World"

michalpokusa, 2023-04-25:
I am not entirely sure what you mean, but there is already a very simple example:
https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/main/examples/httpserver_simple_serve.py

If you delete all docstrings, prints etc. it is only 4-5 lines long, not considering connecting to WiFi. If you want it without a handler for "/", it is 2 lines long. 

One could for sure save some characters but this would not be a realistic scenario for using this lib.
Correct me if I understood this Issue incorrectly.

jrrickerson, 2023-04-25:
I did see the simple server example, but I thought I could provide another that required a bit less work by the end user, and fit the library standard of having a "<library>_simpletest.py" module in the examples folder.  I've submitted a PR for it but let me know if you think it's redundant.

jrrickerson, 2023-04-25:
@michalpokusa Thanks, I appreciate the feedback.  I can make adjustments to the PR to merge the two examples then.

jrrickerson, 2023-04-25:
@michalpokusa I've merged the two examples, and renamed the "simple_serve" example to "simpletest" to adhere to the convention which appears to be in the other CircuitPython libraries and this contributing guide:
https://circuitpython.org/contributing/library-infrastructure-issues

Please let me know if you see any other areas for improvement.

PR #51 (2023-04-25): Add a bare minimum simpletest example
Create the simplest of simple tests for the HTTPServer by just returning a static plaintext string.
The user does not need to create a directory or HTML file or anything.
Fixes #50 

NOTE:  This has not yet been tested on hardware due to lack of availability.
closed 2023-04-25 by FoamyGuy