adafruit/Adafruit_CircuitPython_HTTPServer #60: Setting the hostname - Pico W

adrianblakey, 2023-07-02:
The default hostname is PicoW.

How do I set it to something different?

I am using circuit python 8.2
If I do this:

```
...
from adafruit_httpserver import Server, Request, FileResponse, MIMETypes, GET, JSONResponse
wifi.radio.hostname = str("xx")

...

server.serve_forever(str(wifi.radio.hostname)) # line 67
...
```

The REPL says:

```
>>> Wi-Fi: No IP | Done | 8.2.0-rc.1                                                                                                                                                    
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.                                                                                                     
code.py output:                                                                                                                                                                         
0;🐍192.168.178.89 | code.py | 8.2.0-rc.1socket_resolve_host() returned -2                                                                                                              
Traceback (most recent call last):                                                                                                                                                      
  File "adafruit_httpserver/server.py", line 138, in _verify_can_start                                                                                                                  
gaierror: (-2, 'Name or service not known')                                                                                                                                             
                                                                                                                                                                                        
The above exception was the direct cause of the following exception:                                                                                                                    
                                                                                                                                                                                        
Traceback (most recent call last):                                                                                                                                                      
  File "code.py", line 67, in <module>                                                                                                                                                  
  File "adafruit_httpserver/server.py", line 151, in serve_forever                                                                                                                      
  File "adafruit_httpserver/server.py", line 170, in start                                                                                                                              
  File "adafruit_httpserver/server.py", line 140, in _verify_can_start                                                                                                                  
RuntimeError: Cannot start server on xx:80                                                                                                                                              
0;🐍192.168.178.89 | 140@adafruit_httpserver/server.py RuntimeError | 8.2.0-rc.1                                                                                                        
Code done running.                                                                                                                                                                      
0;🐍192.168.178.89 | REPL | 8.2.0-rc.1                                                                                                                                                  
Adafruit CircuitPython 8.2.0-rc.1 on 2023-06-27; Raspberry Pi Pico W with rp2040
```           

Needless to say removing this and rebooting ...

```
0;🐍Wi-Fi: Authentication failure | REPL | 8.2.0-rc.1                                                                                                                                   
Adafruit CircuitPython 8.2.0-rc.1 on 2023-06-27; Raspberry Pi Pico W with rp2040                                                                                                        
>>>                                                                                                                                                                                     
>>> Wi-Fi: No IP | Done | 8.2.0-rc.1                                                                                                                                                    
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.                                                                                                     
code.py output:                                                                                                                                                                         
0;🐍192.168.178.89 | code.py | 8.2.0-rc.1Started development server on http://PicoW:80        
```

michalpokusa, 2023-07-02:
Instead of `server.serve_forever(str(wifi.radio.hostname))` try `server.serve_forever(str(wifi.radio.ipv4_address))`, hostname will be visible, but if you want to access server via it you need to [configure a mDNS](https://docs.circuitpython.org/projects/httpserver/en/latest/examples.html#server-with-mdns).

adrianblakey, 2023-07-02:
TY

closed 2023-07-02 by dhalbert