adafruit/Adafruit_CircuitPython_HTTPServer #61: mdns setting error - trying to set a custom hostname

adrianblakey, 2023-07-02:
This code:

...
mdns_server = mdns.Server(wifi.radio)
mdns_server.hostname = "cont"
mdns_server.advertise_service(service_type="_http", protocol="_tcp", port=80)
...

Gives this error:

```
0;🐍192.168.178.89 | code.py | 8.2.0-rc.1Traceback (most recent call last):                                                                                                             
  File "code.py", line 29, in <module>                                                                                                                                                  
RuntimeError: Out of MDNS service slots                                                                                                                                                 
0;🐍192.168.178.89 | 29@code.py RuntimeError | 8.2.0-rc.1                                                                                                                               
Code done running.  
```
Is this because I need write access to my local network's DNS server?
If this worked will it allow me to set a custom hostname different from PicoW?

anecdata, 2023-07-02:
Does it work if you turn off web workflow?

adrianblakey, 2023-07-03:
Please can you tell me how I do that?

michalpokusa, 2023-07-03:
About Web Workflow:
https://docs.circuitpython.org/en/latest/docs/workflows.html#web

Try connecting to WiFi manually, as shown in Simple Test (https://docs.circuitpython.org/en/latest/docs/workflows.html#web).

anecdata, 2023-07-10:
@adrianblakey Can this be closed, or it there an issue here distinct from https://github.com/adafruit/circuitpython/issues/8149 ?