adafruit/Adafruit_CircuitPython_HTTPServer #66: Example for SSL

ams1, 2023-08-02:
Hi,

Thank you for this AWESOME initiative!

Any plan to have an example for SSL http server?

zap8600, 2023-08-07:
I've tried using the example on CircuitPython 8.2.0, using both version 4.2.0 and 2.3.0, on an Adafruit Feather ESP32-S2. The error I get is the same on both HTTP server versions. The error below is from my attempt to use version 2.3.0, modified to print the socket type on server start. Here is the error I'm getting.
```
Traceback (most recent call last):
  File "code.py", line 79, in <module>
  File "code.py", line 49, in main
  File "/lib/adafruit_httpserver/server.py", line 137, in poll
RuntimeError: Invalid socket for TLS
```
Here is the line the error is happening at.
```python
            conn, client_address = self._sock.accept()
```
Hope this helps!

anecdata, 2023-08-07:
What are you using for a client? `curl -v` might yield some clues.

michalpokusa, 2023-08-07:
During my testing I was getting the exact same error. I was testing using `curl`, VSCode extension Thunder Client, a script in Python with `requests`, Google Chrome and even other microcontroller (ESP32-S2 TFT) with `adafruit_requests`.

I believe it is something in the CP itself. The interesting thing is that `adafruit_requests` as far as I know can make requests to HTTPS endpoints, so the capability is there.

zap8600, 2023-08-07:
> What are you using for a client? `curl -v` might yield some clues.

I'm using `curl`.

anecdata, 2023-08-07:
The `curl` switch `-v` gives detail about the SSL transaction, I thought it might give a clue where in the process it breaks.

zap8600, 2023-08-07:
> The `curl` switch `-v` gives detail about the SSL transaction, I thought it might give a clue where in the process it breaks.

I haven't looked much into it, but I think it only has to do with CP. Both `curl` and the browser don't get anything back from the microcontroller. They'll throw a connection reset error when the microcontroller resets, though.

anecdata, 2023-08-07:
I loaded up a Pico W and a QT Py S3 (both on CP 8.2.0).

It works on Pico W:
```
% curl -v --insecure https://192.168.6.198
*   Trying 192.168.6.198:443...
* Connected to 192.168.6.198 (192.168.6.198) port 443 (#0)
* ALPN: offers h2,http/1.1
* (304) (OUT), TLS handshake, Client hello (1):
* (304) (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN: server did not agree on a protocol. Uses default.
* Server certificate:
*  subject: CN=picow.local
*  start date: Feb 28 05:42:56 2023 GMT
*  expire date: Feb 28 05:42:56 2024 GMT
*  issuer: CN=picow.local
*  SSL certificate verify result: self signed certificate (18), continuing anyway.
* using HTTP/1.x
> GET / HTTP/1.1
> Host: 192.168.6.198
> User-Agent: curl/7.88.1
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Length: 117
< Content-Type: text/html
< Connection: close
< 
<!DOCTYPE html>
<title>Hello world &middot; Raspberry Pi Pico W</title>
<p>Hello world from Raspberry Pi Pico W!</p>
* Closing connection 0
* TLSv1.2 (OUT), TLS alert, close notify (256):
```
It breaks early in the TLS handshake on S3:
```
% curl -v --insecure https://192.168.6.210
*   Trying 192.168.6.210:443...
* Connected to 192.168.6.210 (192.168.6.210) port 443 (#0)
* ALPN: offers h2,http/1.1
* (304) (OUT), TLS handshake, Client hello (1):
* Recv failure: Connection reset by peer
* LibreSSL/3.3.6: error:02FFF036:system library:func(4095):Connection reset by peer
* Closing connection 0
curl: (35) Recv failure: Connection reset by peer
```
Probably some difference in common-hal, or esp-idf. Perhaps we should move this issue to the `circuitpython` repo.

anecdata, 2023-08-07:
`code.py` passes in `server_side=True` in line 74: `return self._ssl_context.wrap_socket(socket, server_side=True)`
https://docs.circuitpython.org/en/latest/shared-bindings/ssl/index.html#ssl.SSLContext.wrap_socket
`espressif` does nothing with it:
https://github.com/adafruit/circuitpython/blob/6b36bf59e30a4d3d93ec9df1f92ae8adc5fe2094/ports/espressif/common-hal/ssl/SSLContext.c#L40
compared to `raspberrypi`:
https://github.com/adafruit/circuitpython/blob/6b36bf59e30a4d3d93ec9df1f92ae8adc5fe2094/ports/raspberrypi/common-hal/ssl/SSLSocket.c#L166