adafruit/Adafruit_CircuitPython_HTTPServer #68: Submitting empty form, or forms with only checkboxes, causes crash in web server

radair6000, 2023-09-01:
Hi,

I think I found an issue in the web server included in CircuitPython. If you post a form back to the web server that has no input fields, OR contains only chekboxes and they all are clear (which causes no form data to be posted back), the web server seems to crash.

For example this form:
```
<form action="/update" method="post">
    <input type="checkbox" name="test" value="1">
    <input type="submit" value="Update">
</form>
```

When the checkbox is clear, and the form is posted back, when trying to access the field value (or any of the other relevant methods like get(...)):
```
@server.route("/update", POST)
def update(request: Request):
    x = request.form_data.fields

``` 

The web server crashes with this error:
```
  File "adafruit_httpserver/request.py", line 321, in form_data
  File "adafruit_httpserver/request.py", line 144, in __init__
  File "adafruit_httpserver/request.py", line 157, in _parse_x_www_form_urlencoded
ValueError: need more than 1 values to unpack

```
This is using adafruit_httpserver from _adafruit-circuitpython-bundle-8.x-mpy-20230829.zip_ on a Rasberry PI Pico on Circuit Python 8.2.4
If I add a text or hidden input in the form for example, it works:
```
<form action="/update" method="post">
    <input type="checkbox" name="test" value="1">
    <input type="hidden" name="a" value="1">
    <input type="submit" value="Update">
</form>
```

michalpokusa, 2023-09-01:
Hi, thanks for great explaining the source of the issue, I was able to track down the reason and the fix is included in the #67 PR. Thanks!

radair6000, 2023-09-06:
Managed to test this now (sorry for taking so long), and it seems to be working great, thanks!

PR #67 (2023-09-08): 301/302 Redirects, FormData files, Cookies, Token authentication
Despite multiple new features, the overall API itself didn't change much. I believe theese changes, if it comes to that, should be released as 4.x.0 and not 5.0.0.

⭐ Added:

- `Token` authentication class, works the same way as `Bearer`, but with "Token ..." prefix
- Added `MOVED_PERMANENTLY_301` and `FOUND_302` `Status` objects
- `Headers` got new methods: `get_directive()` and `get_parameter()`, `add` and `set`
- `Request.cookies` for retrieving and `cookies` parameter for setting in all `Response`-like classes
- Added docs an [... cut]
commit f2773cd1 (2023-09-08): Fix #68: Incorrect parsing of x-www-from-urlencoded in some cases
closed 2023-09-08 by FoamyGuy