adafruit/Adafruit_CircuitPython_HTTPServer #81: Simple Test (examples/httpserver_simpletest_auto.py)  error  on  M5stack Timer Camera X

sina2, 2024-02-08:
Adafruit CircuitPython 8.2.9 on M5stack Timer Camera X.
I am building simple HTTP server.
I ran examples/httpserver_simpletest_auto.py( https://docs.circuitpython.org/projects/httpserver/en/latest/examples.html#id4 ) and got the following error.
AttributeError : 'SocketPool' object has no attribure 'SOL_SOCET' .
[image]

I saw adafruitt_server/server.py line216 .
        self.host, self.port = host, port

host is "192.168.10.116". port is 80.

Let me know if there is anything else to do.

michalpokusa, 2024-02-08:
Hi, thanks for posting the issue.

The problem you are facing only exists in version 4.5.3, as I made a simple mistake, which resulted exactly in this problem.

Please update your version of the lib to, as yesterday it was fixed and now should be working fine.

PR that fixes the issue described above: https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/pull/80

sina2, 2024-02-08:
I upgraded to version 4.5.4 and the problem is fixed. Thank you!

closed 2024-02-08 by sina2