adafruit/Adafruit_CircuitPython_HTTPServer #92: HTTPServer stops working on CircuitPython version >= 9.0

gmos, 2024-03-28:
HTTPServer stops working on CircuitPython version >= 9.0

Around line 207 in server.py it is assumed that `sock.setsockopt(socket_source.SOL_SOCKET, socket_source.SO_REUSEADDR, 1)` would work as of CircuitPython version 9.0.  Unfortunately it doesn't and raises an `OSError: [Errno 95] EOPNOTSUPP`.

Perhaps better in the circuitpython implementations to just try to set SO_REUSEADDR and catch + ignore the error.

dhalbert, 2024-03-28:
Which board? This is fixed in 9.0.1 for Pico W.

michalpokusa, 2024-03-28:
Duplicate of #90?

gmos, 2024-03-29:
Indeed a duplicate of https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/issues/90 .
And it was on 9.0.0  I'll upgrade to 9.0.1 and retry.

closed 2024-03-29 by gmos