adafruit/Adafruit_CircuitPython_HTTPServer #93: RPI Pico W Circuitpython Webserver on AP

AndreasTheCat, 2024-04-02:
Hello folks (from germany),

[danhalbert](https://forums.adafruit.com/memberlist.php?mode=viewprofile&u=366740) from adafruits forum guided me here, that's why I'm kindly asking here the original [question](https://forums.adafruit.com/viewtopic.php?t=209633).

RPI PICO W should:
* act as an access point
* provide some routes
* do something useful in parallel

Following the example [https://github.com/adafruit/Adafruit_Ci ... asyncio.py](https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/main/examples/httpserver_start_and_poll_asyncio.py) me I hardly tried to spin up the **http-server on wifi.radio.ipv4_address_ap** rather than on wifi.radio.ipv4_address.
Don't get me wrong: clear decorated routes (@server.route...) are best practice. That's why I welcome that pattern.

This led to no success. Indeed this is what's going on:

* I can connect to the AP ---> YES (which is 192.168.4.1)
* the client gets a IP adress assigned ---> YES (which is 192.168.4.XXX)
* the client can fetch http://192.168.4.1/ --> NO

What's wrong here? Any suggestions are appreciated.

```
import time
import board
import digitalio
import wifi
import socketpool
from adafruit_httpserver import Server, REQUEST_HANDLED_RESPONSE_SENT, Request, Response
from asyncio import create_task, gather, run, sleep as async_sleep

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT

ap_ssid = "myAP"
ap_password = "12345678"
wifi.radio.start_ap(ssid=ap_ssid, password=ap_password)
print("Access point created with SSID: {}, password: {}".format(ap_ssid, ap_password))
print("My IP address is", wifi.radio.ipv4_address_ap)

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, "/static", debug=False)
server.start(str(wifi.radio.ipv4_address_ap))

@server.route("/")
def base(request: Request):
    return Response(request, "Hello from the CircuitPython Access Point")

async def handle_http_requests():
    while True:
        pool_result = server.poll()
        if pool_result == REQUEST_HANDLED_RESPONSE_SENT:
            # 
            pass
        await async_sleep(0)

async def do_something_useful():
    global i
    while True:
        led.value = 1
        await async_sleep(1)
        led.value = 0
        await async_sleep(1)

async def main():
    await gather(
        create_task(handle_http_requests()),
        create_task(do_something_useful()),
    )

run(main())
```
P.S.:
* "/static" directory exists
* the client is not browser stuff, I used different HTTP-Clients instead to fetch

michalpokusa, 2024-04-02:
From my initail testing:
- your code works on latests `adafruit_httpserver` and CP 9.0.2
- when troubleshooting, `debug=True` might be useful
- RPicoW blinks the LED
- access point starts

My guess is that you use one of the latest `adafruit_httpserver` which chaged default port from 80 to 5000, thus you can't access it using `http://192.168.4.1/`, try `http://192.168.4.1:5000/` or `server.start(str(wifi.radio.ipv4_address_ap), 80)` (only if you do not have Web workflow enabled, because it also uses port 80).

AndreasTheCat, 2024-04-03:
[michalpokusa](https://github.com/michalpokusa),

will do so, thank you very much for the **:5000** hint

AndreasTheCat, 2024-04-03:
[michalpokusa](https://github.com/michalpokusa),

just installed CP9.0.2 and gave `server.start` port 80. 
Workes great, thank you again very much :-)

michalpokusa, 2024-04-03:
I like you usage with AP, I will add similar example to the lib itself as I believe that might be a popular use case when there is no external network. 👍 

Also please close the issue as completed.