adafruit/Adafruit_CircuitPython_HTTPServer #97: Wifi Co-Processor Support?

eshawveratdirigible, 2024-10-25:
Using the wsgi module for the time being, but the performance is a bit slow. 
Using Adafruit RP2040 Adalogger with CircuitPython version: 9.1.4, adafruit_esp32spi version: 7.1.0, adafruit_wsgi version: 2.0.3.
Any suggestions or recommendations for a solution would be appreciated. 

Copying this from an old issue #86 . 

-----------------------------------------------------------------------------------------------
Seeing the same issue and would like to throw my support behind some resolution.

Using the Adafruit RP2040 Adalogger with CircuitPython version: 9.1.4, adafruit_httpserver version 4.5.9, and adafruit_esp32spi version 8.4.2.
Adafruit RP2040 Adalogger and AirLift Wifi FeatherWing Co-Processor are connected via a FeatherWing Tripler.

I got the httpserver started, but cannot access the endpoint from my laptop, but can ping the ip address of the microcontroller from my laptop.
Httpserver started with the addition of wrapper classes:

```python 
import board
import busio
from os import getenv
from digitalio import DigitalInOut
import adafruit_connection_manager
import adafruit_requests as requests
from adafruit_esp32spi import adafruit_esp32spi
from adafruit_esp32spi import adafruit_esp32spi_socketpool as SocketPool
from adafruit_httpserver import Server, Request, Response

secrets = {
    "ssid": getenv("WIFI_SSID"),
    "password": getenv("WIFI_PASSWORD")
}

esp32_cs = DigitalInOut(board.D13)
esp32_ready = DigitalInOut(board.D11)
esp32_reset = DigitalInOut(board.D12)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

pool = adafruit_connection_manager.get_radio_socketpool(esp)
ssl_context = adafruit_connection_manager.get_radio_ssl_context(esp)
requests = requests.Session(pool, ssl_context)

esp.connect_AP(secrets["ssid"], secrets["password"])

pool = SocketPool.SocketPool(esp)

old_socket = SocketPool.Socket(pool)

class socket:
    def __init__(self, AF_INET, SOCK_STREAM):
        self.AF_INET = AF_INET
        self.SOCK_STREAM = SOCK_STREAM

    def __call__(self, AF_INET, SOCK_STREAM):
        return self

    def setsockopt(self, level: int, optname: int, value: int):
        return (level, optname, value)

    def bind(self, values: tuple):
        return (values)

    def listen(self, value: int):
        return value

    def setblocking(self, value: bool):
        return value

    def settimeout(self, value: float):
        return value

    def accept(self):
        conn = my_socket
        sock_num = 1
        sock = old_socket._interface.get_remote_data(sock_num)
        addr = self.bind((sock['ip_addr'], 80))
        return (conn, addr)

    def recv_into(self, buffer: bytearray, nbytes: int = 0):
        return old_socket.recv_into(buffer, nbytes)

    def close(self):
        return True

class My_Socket_Pool:
    def __init__(self, pool, socket):
        self.pool = pool
        self.socket = socket

    def getaddrinfo(self, host, port):
        return (host, port)

    def AF_INET(self):
        return self.pool.AF_INET

    def SOCK_STREAM(self):
        return self.pool.SOCK_STREAM

    def SOL_SOCKET(self):
        return 1

    def SO_REUSEADDR(self):
        return 4

my_socket = socket(pool.AF_INET, pool.SOCK_STREAM)
my_socket_pool = My_Socket_Pool(pool, my_socket)

server = Server(my_socket_pool, "/static", debug=True)

@server.route("/test")
def base(request: Request):
    return Response(request, "test worked")

server.serve_forever(str(esp.pretty_ip(esp.ip_address)), 80)
```

anecdata, 2024-10-25:
@eshawveratdirigible Given Dan's comment in #86, I think it makes more sense for an issue in ESP32SPI, rather than a duplicate open issue here. Related: https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/issues/203

eshawveratdirigible, 2024-10-26:
Thanks for your quick response. This issue can be closed. I'll look for solutions in the other library repository.

closed 2024-10-26 by dhalbert