adafruit/Adafruit_CircuitPython_HTTPServer #99: mDNS not working

riancz, 2024-11-16:
Device: Raspberry Pi Pico

mDNS seems to not work when combining the two basic examples from the project: [Manual AP](https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/main/examples/httpserver_simpletest_manual_ap.py) & [mDNS](examples/httpserver_mdns.py).

I can reach the page when going directly to my ip address on port 5000, however when trying to reach it on abcd.local:8080 it doesn't do anything.

```
import wifi
import socketpool
from adafruit_httpserver import Server, Request, Response, FileResponse
import mdns

# Set up access point
print("Creating access point...")
wifi.radio.start_ap(ssid="MyNetwork", password="SomePassword!")
print(f"Created access point {SSID}")

#Set up mDNS
mdns_server = mdns.Server(wifi.radio)
mdns_server.hostname = "abcd"
print("MDNS Hostname: " + mdns_server.hostname)
mdns_server.advertise_service(service_type="_http", protocol="_tcp", port=8080)

# Set up web server
pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, "/static", debug=True)

@server.route("/", methods=['GET'])
def index(request):
    """ Render the Index page"""
    if request.method == 'GET':
        return FileResponse(request, "index.html", "/static")

server.serve_forever(str(wifi.radio.ipv4_address_ap))
```

Did I do something wrong, or is it some bug?
Thanks

michalpokusa, 2024-11-16:
I cannot test right now, but if this is the exact code you are using, then the http server port does not match the mDNS port, which I believe might be a problem here.

riancz, 2024-11-16:
It doesn't work even if the mDNS port is set to 5000 as well.
`mdns_server.advertise_service(service_type="_http", protocol="_tcp", port=5000)`

Also just to be sure I tries to set up both to port 80, with no success.
`mdns_server.advertise_service(service_type="_http", protocol="_tcp", port=80)`
`server.serve_forever(str(wifi.radio.ipv4_address_ap), port=80)`

anecdata, 2024-11-16:
If I'm reading this right, mDNS seems to be hard-coded to station:
https://github.com/adafruit/circuitpython/blob/8cdd96c2055de8a03446d7bc34785d3a32e4a57e/ports/raspberrypi/common-hal/mdns/Server.c#L41C27-L41C35