adafruit/Adafruit_CircuitPython_HTTPServer #102: SSL import error

bludin, 2025-03-20:
The import statement on line 15 of server.py causes an error on platforms that do not support SSL (eg M4)

bludin, 2025-03-21:
I just entered this on behalf of somebody I helped in the [Adafruit Circuitpython forum](https://forums.adafruit.com/viewtopic.php?t=217397). I proposed a little change to the code there, that would allow for the import and use on non-SSL-enabled platforms.

michalpokusa, 2025-03-23:
Currently, importing the `hashlib` module works in a similar way.

```python
try:
    try:
        import hashlib
    except ImportError:
        import adafruit_hashlib as hashlib
except ImportError:
    print(
        "WARNING: hashlib module not available and adafruit_hashlib not installed.",
        "Websocket support will not work.",
    )
```

Maybe this could be applied for `ssl` too? If there is a use case for `adafruit_httpserver` on boards that do not have `ssl` module, in my opinion it should not throw an error just because it is missing a module it wouldn't use either way.

PR #103 (2025-06-24): Run on platforms without ssl
Catch the errors when importing ssl, and raise an exception in Server init when https is set.
This will allow using httpserver on ESP32SPI with https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI/pull/218
Fixes #102
closed 2025-06-24 by Neradoc